strangeparts/gasmask #3: Design is to huge for most 3d printers

schaurian, 2020-06-15:
This might not be an issue. But it definitely a big disadvantage.  I can't fit it on my Prusa mk3s. Which is a very common 3d printer. Maybe the Part from the filter could be closer to the mouth?
[image]

tytytyty74, 2020-06-16:
I have to agree with you there, on my ender 3 there's a single orientation it will fit, which is very difficult to print, requiring a significant amount of support material, upping the print time to over 60 hours

schaurian, 2020-06-16:
> I have to agree with you there, on my ender 3 there's a single orientation it will fit, which is very difficult to print, requiring a significant amount of support material, upping the print time to over 60 hours

Do you have and idea how to improve it. I have some skills in fusion 360. Ideally it would be easy to print on any printer without any support material.

stephen304, 2020-06-17:
Just to bridge some of the discord discussion to here, it seems the direction we are moving is to separate the canister from the mask and use NATO 40mm threads so that the mask can be compatible with standard canisters and a wide variety of adapters to practically any other canister, also allowing the custom canister to be removed without opening up the contents. That will remove the top of the canister completely from the model which should really help it fit on more print beds. There are some experienced cadders who dropped into discord so we should coordinate to make sure no work is duplicated. Somebody also made a fusion360 cloud team so collaboration will probably happen there, maybe we can sync it back here periodically.

Also we don't have the female nato 40mm threads yet, but we successfully printed the male threads and verified it with a commercial mask.

stephen304, 2020-07-04:
@schaurian A modified mask is now in the masks folder if you'd like to check that out. It should be easier to fit it on a prusa