strangeparts/gasmask #4: Exhalation valve and dead space

skaihoj, 2020-06-15:
I know hhis has been mentioned elsewhere, but this seems like the place to stpre and organize ideas.

The device needs an exhalation valve. It needs to be located on the other side if the "inner mask", and the inner mask should not seal against the face. It should preferably be made such that the fresh air arrives on the outside of the inner mask, and is exhaled inside it. 

Dead space should be minimised. This is the volume of gas that gets inhaled again after being exhaled. The current design has so much dead space, that it might cause carbon dioxide poisoning. Much of this can be removed with a carefully placed exhalation valve.

Liam-Stark, 2020-06-15:
Would two check valves, placed on either side of the main valve, be a reasonable solution? Is there a better place for exhalation valve(s)?

Also, can you elaborate on the third sentence? Maybe a diagram of the suggested airflow? I just had a hard time following lol.

skaihoj, 2020-06-15:
The filter will usually have high resistance to airflow, so only one check valve is needed, and no other valves of any kind. 

If you are breahing into something like a snorkel, some of the air you exhale will be left in the snorkel. When you inhale again, this sir goes back in your lungs. If the volume of such a snorkel is large enough, no new air will enter your lungs. if the same air gets used again and again, you will suffocate. See also https://en.wikipedia.org/wiki/Dead_space_(physiology)#Mechanical_dead_space

In the current design, the volume of the mask and the filter both contribute to this dead space.  Adding an exhalation valve will remove the filter from the volume getting breathed back in.

stephen304, 2020-06-17:
There have been a couple valve design ideas going around the discord, complexity and materials that may be hard for some people to find can be a problem with using 1 specific valve design, so we have the idea of using some standard threaded hole that we can design a plug for, with our valves embedded in it. That way assemblers can choose an appropriate exhale valve based on their requirements and available materials.

As far as I can tell these are the remaining tasks to resolve this:
* Decide on a location for the vent port
  * I think we all agree that it should vent from the inner mask.
  * If we add a new inlet to the side of the mask (which would intake air to the larger space, maybe this would help reduce fogging?) we could reuse the existing port location for the vent valve
  * We could also add a second port above or below the current one and use one as in and one as out - I'm not sure whether you would want a tube to get the intake into the larger space rather than the mouth area
* Decide on a threading for the vent port
  * I initially wanted to just use nato since we're already working with it, but some suggested that making it smaller/different would help reduce mistakes. I don't really have an opinion on this as long as everyone agrees
* Add a channel to the outside of the vent port to reduce the chance of contamination ingress via backflow.
  * This is my idea to make the mask more resistant to less perfect valve designs. My fully 3D printed one way valve does allow a small amount of air to ingress before the valve fully closes. I have to test to determine whether it's just air inside the valve or whether any air outside the valve actually gets in, but I think adding this feature will be good to enable a wider variety of valve designs. Probably this will happen last and may not be necessary - we'll see :)

If you haven't joined the discord, a lot of discussion is going on there. Collaboration will probably happen in fusion360 cloud, and maybe changes will be synced back here periodically.

skaihoj, 2020-06-18:
I am on the discord, but i find the discussion there hard to follow, since there is only one room there for discussing the gas mask. all the discussion going on in one place makes it very time consuming to participate. 

* We should only need one check valve, if it works properly. The resistance to airflow in the filter should be enough to prevent backflow. 
* The threading for the vent port should mainly be decided by what will actually work on a 3D printer. 3D printing threads that last can be hard.

stephen304, 2020-06-18:
@skaihoj Yeah it's been a lot of chatter in there. Good to know about the valves. For the thread, a lot of threadings print pretty well, even down to small M sizes. The NATO threads are actually pretty huge, so there's no problem with the filter port. Our test pieces so far have worked with off the shelf NATO masks. If we chose a different thread for the vent port then it will likely be a lot finer of a thread. I don't really have a strong opinion on the vent port thread so I'll probably just defer to those with engineering experience.

stephen304, 2020-07-04:
I believe this is now resolved as my one way valve has been tested and the masks have been redesigned to accept a threaded valve assembly. We can probably close this issue and any tangential concerns can go in a new issue.

skaihoj, 2020-07-06:
That seems reasonable. From what i've seen on the discord, you seem to have any issues raised here under control.

closed 2020-07-06 by skaihoj