strangeparts/gasmask #5: Exhaust outlet needed

Cyrko, 2020-06-15:
Hi ! 

Thanks for your project ! It seems very promising however I think there needs to be an exhaust port on the mask. Since it is difficult to breathe through the filter, it would be greatly beneficial to expire effortlessly. It would also help to manage the humidity in the filter. I am speculating but I guess that a damp one would be less effective since the activated charcoal would be clogged up.

Here are some reference picture from a mask I used during my military service: 

[image]
Behind the "grate" is a voice diaphragm. The two lower oppenings are the inlets and outlets. Each one of them houses a one way valve. 

[image]
This is a close-up of the outlet

[image]
The gray opening is just the voice diaphragm.
[image]
[image]
[image]
[image]
The rubber of the valve is pretty thin so it may be possible to use balloons to make membranes. Their curved shape may be used to ensure a tight contact and mimic the bevel of the original one. 
[image]

Regarding the nose and mouth separate area, it appears that its primary function is to unfog the lenses using fresh air. Since the eyes are extremely sensitive it makes sense to have only one effective seal around the mask. 
[image]
[image]
[image]
[image]

Thanks again for your work and I hope to be able to help you further !

3devangelist, 2020-06-17:
I have made one for covid, to test out materials and use TPU, 
https://wikifactory.com/+movimentomakerpt/diy-paper-mask

stephen304, 2020-06-17:
I made a lengthy comment on the previous issue about vent ports and some of the stuff we've been discussing in the discord channel. But I'll add a few comments based on what you've suggested:

* What are your thoughts on the importance of valves on both the filter and outlet? I think so far we've been discussing having just an exhale valve to prevent air intake when breathing in. When exhaling, the pressure difference should be relatively low if the vent port is working, but some humid air may go back through the filter. I'm not too knowledgeable about this
* For your balloon idea, would you like to design or spec out a valve based on this idea? We have a couple valve ideas in development, and plan on making them threaded so assemblers can choose a valve based on requirements and materials they have available. So if you could assist with a prototype using a balloon, we could add it to the options
* I'm not sure I completely understand the last part about airflow between the 2 sections of the mask - is the design of your mask so that air intake comes into the eye area and leaves from the mouth area? And is there a vent hole to allow free flow of air between the 2 chambers or do you think another one way valve would help to control the flow of humid air from exhaling reaching the eye area?

justinl681, 2020-06-18:
Hello! The design team and I are working on a one-way valve. If you'd like, check my pinned github page and give feedback!

stephen304, 2020-07-02:
There are now a couple of valve designs, one of which is tested, so I would suggest closing this issue and creating new ones for any remaining concerns

closed 2020-07-02 by Cyrko